pvvx/ZigbeeTLc #188: Brackets for Smiley do not disappear

web2brain, 2025-04-15:
I updated the first of a few MHO-C401N from version 1.2.4 to version 1.2.6 and saw afterwards, that the brackets for the smiley do not disappear any more when the smiley has been deactivated.

Left version 1.2.6 with smiley deactivated, right version 1.2.4 with smiley deactivated:

[image]

I think it looks nicer when the brackets are disabled as well, but of course it is not an important bug.

pvvx, 2025-04-15:
https://github.com/znanev/MHO-C401?tab=readme-ov-file#segment-165

pvvx, 2025-04-15:
Take another look at the diagram of the E-ink panel segments in the MHO-C401 at https://github.com/znanev/MHO-C401?tab=readme-ov-file#segment-165.
These segments on the MHO-C401 are connected to one wire in the E-Ink:
1. Degree symbol and common parts of C and F
2. Decimal point
3. Side of faces
4. Percent sign

MHO-C401N (N is the new variant) has a different E-Ink.

juha63, 2025-04-16:
epd_mho_c401n.c:

`void show_smiley(u8 state){
        // off
        scr.display_buff[5] &= ~(BIT(2) | BIT(4) | BIT(6)); // do not reset %
        scr.display_buff[6] = 0;
        **scr.display_buff[7] &= ~(BIT(4));**
        if (state) {
                **scr.display_buff[7] |= BIT(0); /* (  ) */**
        }
`

The brackets (buffer 7 bit 0) will be set if state != 0, but won't be cleared if state will be switched to off.

I don't know why you turn off buffer 7 bit 4, because this segment has nothing to do with the smiley? May be a typo?

pvvx, 2025-04-16:
Firmware and code "smiley" for MHO-C401N updated.

> I don't know why you turn off buffer 7 bit 4, because this segment has nothing to do with the smiley? May be a typo?

These are obvious errors... The display options for "smiley" have been corrected to another option.

juha63, 2025-04-17:
@pvvx: Works perfectly, thank you! 🥇

web2brain, 2025-04-17:
Likewise for me, thanks @pvvx 
I'll close the isses as it has been resolved.

closed 2025-04-17 by web2brain