pvvx/ZigbeeTLc #191: ZTH05 incorrect model id

juha63, 2025-04-22:
Hi Victor,

thanks for your great work supporting all these cheap sensors!

In board_zth05.h you set

`#define ZCL_BASIC_MODEL_ID        {6,'T','H','0','5','-','z'} // ZTH05`

so the device is showing up as _TH05-z_

Would you mind setting this to 

`#define ZCL_BASIC_MODEL_ID        {7,'Z','T','H','0','5','-','z'} // ZTH05`

Thanks
Jürgen

pvvx, 2025-04-22:
No. I can't because it's already used in ZHA and Z2M. Changing the device name would require changes to a lot of third party software.

/zhaquirks/custom/telink.py: 
```
    QuirkBuilder("MiaoMiaoCe", "MHO-C401N-z")
    .applies_to("MiaMiaoCe", "MHO-C401N-z")  # typo until v. 1.2.2
    .applies_to("MiaoMiaoCe", "MHO-C401-z")
    .applies_to("MiaoMiaoCe", "MHO-C122-z")
    .applies_to("MiaMiaoCe", "MHO-C122-z")  # typo until v. 1.2.2
    .applies_to("Xiaomi", "LYWSD03MMC-z")
    .applies_to("Tuya", "TS0201-z")
    .applies_to("Tuya", "TH03Z-z")
    .applies_to("Tuya", "ZTH01-z")
    .applies_to("Tuya", "ZTH02-z")
    .applies_to("Tuya", "LKTMZL02-z")
    .applies_to("Tuya", "ZY-ZTH02-z")
    .applies_to("Sonoff", "TH03-z")
    .applies_to("Qingping", "CGDK2-z")
```

closed 2025-04-22 by juha63

pvvx, 2025-04-22:
@juha63 - All firmwares have the ability to change the name and manufacturer of the device by the user.
Write the attribute `model` (id: 0x0005) or `manufacturer` (id: 0x0004) in the cluster
Basic (Id: 0x0000) to whichever you like.

https://github.com/pvvx/ZigbeeTLc/issues/84

juha63, 2025-04-22:
I know and I did that in my installation already. Just thought to save myself these steps. ;-)