pvvx/ZigbeeTLc #193: Xiaomi LYWSD03MMC-z - spikes after upgrade to 3001-0126

dgrazvan, 2025-04-22:
After upgrade over OTA from 3001-0123 to 3001-0126 spikes appeared as in the image (blue). Yellow is a sensor located in the same area (<20cm). The spikes appear randomly. I haven't found a repeating pattern.

[image]

Zigbee2MQTT version
2.2.1

Adapter firmware version
8.0.2 [GA]

Adapter
EmberZNet SLZB-06M

Setup
Add-on on Home Assistant - virtual machine on proxmox

dgrazvan, 2025-04-22:
More spikes

[image]

pvvx, 2025-04-23:
Spikes, spikes! :) :)
Two thermometers in 20 cm.
One higher on the wall. Steam from the teapot will fall on him.

[image]

dgrazvan, 2025-04-23:
I understand the situations and agree with them, but in my case I think the only one that would apply could be breathing, but it also happens when there is no one in the room.
The bigger question is why this behavior after I upgraded. This does not happen with version 3001-0123. 
I have the 2 thermometers placed on a sideboard like in the picture.

[image]

pvvx, 2025-04-23:
To obtain a stable temperature graph, the thermometer should be placed in a sealed foam box. At the very least, in a cardboard box. And put it in a place where there is no sunlight. For example, closed in a closet. In all other cases, fluctuations from air convection will be observed. 
The readings of two thermometers on the table may differ by several degrees due to convection. 
From infrared light - up to ten degrees.

closed 2025-04-23 by pvvx

dgrazvan, 2025-04-23:
The graph zize in image is about 1.2gr C. Thermometers do not stand in the sun or in air conditioning.
I wanted to report a change in behavior after upgrading from v3001-0123 to 3001-0126. There were no other changes in position, environment, etc. Just the zigbee firmware version.
If this is an attitude problem you can close the case. I thought I could help by reporting a possible problem. I can manage to revert to the previous version or choose other sensors. 
Have a nice day

pvvx, 2025-04-23:
Xiaomi LYWSD03MMC has several types of HW. You also did not specify which version you have.

dgrazvan, 2025-04-23:
I'm not a telepath either, to know what details are needed. If more details were needed, I think it would have been more professional to ask. I described it from my point of view as a user, and I gave what details I thought were necessary.
Anyway. It doesn't matter to me anymore. For me I solved the problem and the spikes disappeared on the same conditions.
Have a nice day. Good job.