pvvx/ZigbeeTLc #194: 0.1.2.6 + Z2M - Battery draining really fast (2x AAA)

VivantSenior, 2025-04-24:
Hi,

I'm using zigbee firmware together with [2x AAA mod](https://www.printables.com/model/1098300-yet-another-aaa-battery-case-for-xiaomi-bluetoothz):
[image]

It worked perfect on 0.1.2.4 (tested for a month). After upgrade to 0.1.2.5 all of my 11 Xiaomi LYWSD03MMC stopped to report values, dropped from the network and were unable to rejoin Z2M network. Therefor I uploaded Bluetooth ATC firmware (v5.1) using [USB2COM](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html) to most of them (as 3-4 rejoin the network after some time and I was able to update them from Z2M OTA) and later uploaded newest 0.1.2.6 firmware using [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html). Now new 2x AAA are working for 2-3 days and that's it. Out of 11 sensors 3 are working as expected (the same batteries as in the 8 that died).

Reference: #179, #181.

pvvx, 2025-04-24:
Xiaomi LYWSD03MMC B1.4 with an old CR2032 battery has been running on v0.1.2.6 for two weeks. No increase in consumption has been noticed.
When modifying firmware, I always check consumption on PowerProfiler.

Specify:
1. What is the HW (B1.4..B2.0) version of the thermometers with dead batteries
2. Were the working thermometers updated via ZigbeeOTA or a programmer?

guillermohor, 2025-04-25:
That happens with 3 of my 8 sensors. Before with a litium cr2032 battery was like more than 1 month. Now they only last 4 days.

juha63, 2025-04-25:
I had the same: Massive battery drain after update to 1.2.6 on all my sensors. What I didn't realize was, that my wifi channel switched automatically the same time (thats another story and not related to this update!) to a channel, that interfered with the zigbee channel.

I switched back the wifi channel -> no battery drain anymore.

May be this helps.

Damianod27, 2025-04-25:
Hello. I’m joining the discussion – I have exactly the same issue. Previously, all my sensors were running on firmware version 0.1.2.4 and everything worked fine until I updated them to version 0.1.2.6. Following the recommendations, I removed them from the Zigbee gateway and re-paired them after the update. I performed the update using Zigbee OTA.
I power the sensors with 16340 rechargeable batteries with a capacity of 650–700 mAh and a voltage of 3V. Unfortunately, the batteries drain within 2–3 days. In some cases, a protection circuit against excessive or improper power draw is triggered – the batteries shut down and stop supplying power. To deactivate this protection, the battery needs to be connected to a charger for about 1–2 seconds – after that, the voltage is restored and reaches 2.95–3V. Regards

[image]

[image]

[image]

[image]

pvvx, 2025-04-25:
https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#chipset-lywsd03mmc-hwb14

_For LYWSD03MMC, the HW version is determined by the display and sensor addresses on I2C:_

HW | LCD I2C addr | SHTxxx I2C addr | Note
-- | -- | -- | --
B1.4 | 0x3C | 0x70 (SHTC3) |  
B1.5 | UART | 0x70 (SHTC3) |  
B1.6 | UART | 0x44 (SHT4x) |  
B1.7 | 0x3C | 0x44 (SHT4x) | Test original string HW
B1.9 | 0x3E | 0x44 (SHT4x) |  
B2.0 | 0x3C | 0x44 (SHT4x) | Test original string HW

_Version 1.7 or 2.0 is determined at first run by reading the HW line written in Flash. Display matrices or controllers are different for all versions, except B1.7 = B2.0._

[image]

Table: [ID numbers of alternative firmware](https://github.com/pvvx/pvvx.github.io/blob/master/README.md#id-numbers-of-alternative-firmware)

Damianod27, 2025-04-25:
@pvvx Yes I saw it but I didn't find how to read it. I use Zigbee2Mqtt and I don't see such an option

pvvx, 2025-04-25:
> I power the sensors with 16340 rechargeable batteries

From 3.7V the chip degrades.
After working at increased voltage, the chip will always consume more, but work. This is with small increases in voltage over 3.6 V. Increased consumption will be permanent. Only replacing the chip will help.
If the increase was more than 4 V and for a long time, then something burns out in the chip.
Tested on several options.

juha63, 2025-04-25:
@Damianod27 

Switch to the device page. Then tab "Dev console". Use "Endpoint 1, cluster gen_basic, attribute hwVersion" and press READ. 

The hwVersion will be shown soon after.

[image]

juha63, 2025-04-25:
`#define BOARD_LYWSD03MMC_B14		0 `

B1.4

Damianod27, 2025-04-25:
@pvvx I have 8 sensors. Each of them is powered by a 3V EverActive battery. I have a problem with 6 sensors. 4 are powered by EverActive 650 series batteries and they discharge to the point of power cut off and 2 are powered by EverActive 700 series and they block and do not supply power.

Damianod27, 2025-04-25:
@pvvx I checked the other 5 sensors. Each is 1.4. It's a bit strange that 6 have a problem and 2 work properly. I only remember that the ones with problems had the original software version 1.xx after purchase and the ones that don't have a problem had 2.xx. Maybe that matters.

juha63, 2025-04-25:
@pvvx 

> > [@pvvx](https://github.com/pvvx) Thank you very much. I currently have 3 working. The other 5 batteries are flat. What does 0 mean?
> 
> Table: [ID numbers of alternative firmware](https://github.com/pvvx/pvvx.github.io/blob/master/README.md#id-numbers-of-alternative-firmware) -> HW ID

BTW an off topic: One of my LYWSD03MMC shows hwVersion "15" but I can't find this in the definitions???

Damianod27, 2025-04-25:
> > [@pvvx](https://github.com/pvvx) Dziękuję bardzo. Obecnie mam 3 działające. Pozostałe 5 baterii jest rozładowanych. Co oznacza 0?
> 
> [Xiaomi LYWSD03MMC B1.4 ze starą baterią CR2032 działa na wersji v0.1.2.6 od dwóch tygodni.](https://github.com/pvvx/ZigbeeTLc/issues/194#issuecomment-2827246698)
> 
> Wszystko zostało z nimi przetestowane i PowerProfiler wskazuje średnie zużycie mniejsze niż 14 µA przy ustawieniu interwału pomiaru na 30 sekund.
> 
> > Mam 8 czujników. Każdy z nich jest zasilany baterią 3V EverActive. Mam problem z 6 czujnikami. 4 są zasilane bateriami EverActive serii 650 i rozładowują się do punktu odcięcia zasilania, a 2 są zasilane bateriami EverActive serii 700 i blokują się i nie dostarczają zasilania.
> 
> Program sprawdza napięcie zasilania i jeśli jest ono niższe niż 2,0 V, termometr przechodzi w tryb głębokiego uśpienia z poborem prądu mniejszym niż kilka μA. W tym trybie bateria 100 mAh o napięciu 2,0 V rozładuje się dopiero po kilku latach.

Only the voltage of 0.6V is only after blocking and activation of the protections. After applying power to the battery, i.e. attempting to charge for 1-2 seconds, the voltage returns to 3V. I transferred these batteries to the Fibaro Motion Sensor Z-Wave for a few days and there is no problem with them, just like it was on the 0124 software.

pvvx, 2025-04-25:
> BTW an off topic: One of my LYWSD03MMC shows hwVersion "15" but I can't find this in the definitions???

Is this an old firmware version? Then it is B1.5

Damianod27, 2025-04-25:
@pvvx Can they now be reprogrammed via Telink? Telink currently does not detect these sensors via Bluetooth.

juha63, 2025-04-25:
> > BTW an off topic: One of my LYWSD03MMC shows hwVersion "15" but I can't find this in the definitions???
> 
> Is this an old firmware version? Then it is B1.5

Yes, it is 1.2.4. Thanks!

guillermohor, 2025-04-25:
It would be great to be able to revert to a previous firmware version without a hardware UART flasher

Damianod27, 2025-04-25:
> Świetnie byłoby móc powrócić do poprzedniej wersji oprogramowania układowego bez konieczności flashowania sprzętowego UART

Yes I agree with that. There is a lot of work via UART. Soldering etc. and in my case 8 sensors. It is a pity that I changed the 0124 software which worked perfectly for me. I see that there is already version 0127 but it does not show up in the Zigbee2MQTT OTA

pvvx, 2025-04-25:
Maybe it would be great to delete this repository altogether?
This is instead of answering what version of the HW thermometer you have? And like let the rest and new firmwares continue to have the error for others?

Checking new firmware versions was transferred to users a month ago.
I am not able to check more than 150 variants supported by HW and FW alone. A full check requires having all the software used by users and testing for all functions. 

> I see that there is already version 0127 but it does not show up in the Zigbee2MQTT OTA

I don't post versions in the general list for z2m at all. Someone else did it and it's not a fact that it has the correct firmware.

Damianod27, 2025-04-25:
> Maybe it would be great to delete this repository altogether? This is instead of answering what version of the HW thermometer you have? And like let the rest and new firmwares continue to have the error for others?
> 
> Checking new firmware versions was transferred to users a month ago. I am not able to check more than 150 variants supported by HW and FW alone. A full check requires having all the software used by users and testing for all functions.
> 
> The latest changes in the new FW were made because of bugs in ZHA and Z2M. To work around them. ZHA and Z2M do not want to comply with the Zigbee 3.0 standards and do not fix their software. It is possible that Z2M made some changes, for example, disabled switching to LongPoll... Then there will be increased consumption. Version 0.1.2.5 with a LongPollControl period of 60 seconds did not work at all. At the same time, some other Zigbee devices communicate with pauses of hours...
> 
> > I see that there is already version 0127 but it does not show up in the Zigbee2MQTT OTA
> 
> I don't post versions in the general list for z2m at all. Someone else did it and it's not a fact that it has the correct firmware.
> 
> Firmware and "index.json" are located in https://github.com/pvvx/ZigbeeTLc/tree/master/bin
> 
> How to make Zigbee-OTA into an old version or through an intermediate BLE installation via Zigbee-OTA has been described many times.

@pvvx At no point did I say that I have any complaints towards you or anyone else. I only mentioned that with firmware version 0124, everything worked perfectly for a long time, and after updating to 0126, issues started from day one — and as we can see, I’m not the only one experiencing this. It was my decision to update to 0126, so if I have any complaints, they are only toward myself.

I just wanted to join the already reported issue, nothing more. Thank you for maintaining this repository. Please also note that I already thanked you and apologized for my lack of knowledge — and for the time you had to spend explaining things to me.

So I don’t understand where you got the impression that I was complaining. On the contrary — I have great respect for what you do, the knowledge you have, and the patience you show toward people who don’t have the same level of experience. You have to understand that not everyone is as familiar with this as you are, and with so many posts, it’s really hard to find specific information. For you it’s second nature; for others, it’s uncharted territory.

Have a great day!

pvvx, 2025-04-25:
There is no documentation on the EverActive 650 and 700 batteries on the Internet. As a result, its behavior at low or pulse current is unknown, as well as the actual voltage levels it produces.

Damianod27, 2025-04-25:
@pvvx Thank you for your help and time. It is possible, as you write, that the batteries damaged my sensors. I charged the batteries only once right after purchase, so they were never recharged while using Xiaomi sensors. After 4 hours, I found a step-by-step description of how to perform updates via OTA Zigbee2MQTT. Old description and I had to make a few changes to make it work, but it is currently being updated. I will test the 0124 software on one sensor and the 0127 on the other and change the batteries to a Varta CR123A battery. I will let you know in a few days. Best regards

pvvx, 2025-04-25:
Let's take Xiaomi LYWSD03MMC B1.5. It has a completely different sensor and display chip. The consumption in sleep mode is higher than that of HW B1.4.
Firmware 0.1.2.6.

[image]

Sleep current B1.5 - 7.58 μA:

[image]

Average current calculated for 10 minutes: 12.7 μA
Peak (maximum) current during transmission: 8.8 mA.

[image]

And the question is - how could such a current of 13 μA discharge a battery with a capacity of 700 mAh?
700 / 0.013 = 53846 hours of operation

guillermohor, 2025-04-26:
> [@pvvx](https://github.com/pvvx) I checked the other 5 sensors. Each is 1.4. It's a bit strange that 6 have a problem and 2 work properly. I only remember that the ones with problems had the [...]

https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/

I tried this but the update didnt show Up.

pvvx, 2025-04-26:
> https://smarthomescene.com/guides/how-to-setup-zigbee2mqtt-local-ota-index/
> 
> I tried this but the update didnt show Up. i think only ones 1.4 hw 0 have battery drain to me.

To update to ver 0.1.2.7: [index_v0127.json](https://github.com/pvvx/ZigbeeTLc/blob/master/bin/index_v0127.json)
For ZigbeeOTA in BLE version: [index_v51.json](https://github.com/pvvx/ATC_MiThermometer/blob/master/zigbee_ota/index_v51.json)

There is README everywhere:  https://github.com/pvvx/ZigbeeTLc/tree/master/bin#readme , https://github.com/pvvx/ATC_MiThermometer/tree/master/zigbee_ota#readme

You don't need to copy the firmware itself.
You only need to copy and specify the "index_xxx.json" file in the "configuration.yaml" configuration.

After the update, remove the link in "configuration.yaml". Otherwise, some "Tuya" may be caught by the update, which use other people's identifiers.

VivantSenior, 2025-05-02:
> Specify:
> 
> 1. What is the HW (B1.4..B2.0) version of the thermometers with dead batteries?
> 2. Were the working thermometers updated via ZigbeeOTA or a programmer?

1. CHMLmini_TLW_V1.4
    201909
    HW: B1.4
2. I updated them using Z2M OTA feature.

New 2xAAA worked for 2,5 days and dropped from 1,6V to 1,16V each. Now I've put CR2032 to the same thermometer and will see what will happen.

> The capacity of regular AAA batteries is 540 mAh. 3 days = 72 hours. 540/72 = 7.5 mA -> That is, to discharge an AAA battery in 72 hours, a constant load of 7.5 mA is required.
> 
> If suddenly the sleep command does not work, then the working thermometer will constantly consume **up to 4 mA**.
> 
> Measuring the current of the operating processor in Xiaomi LYWSD03MMC B1.4 without "sleep" in the ZigbeeTlc v0.1.2.6 firmware: [image]
> 
> During transmission - more. But if the transmission is always on, and from a dozen thermometers, then the Zigbee network will not work.
> 
> -> AAA batteries discharge will require the CPU to run without "sleep" for a week.

I understand. Is there any other function of the termomethers (beside OTA) that could drain the batteries like constantly rejoining the network?

> [@Damianod27](https://github.com/Damianod27)
> 
> Switch to the device page. Then tab "Dev console". Use "Endpoint 1, cluster gen_basic, attribute hwVersion" and press READ.
> 
> The hwVersion will be shown soon after.

I don't think this approach worked in my case for two sensor I see in the logs:
```
zhc:tz: Read result of 'genBasic': {"hwVersion":0}
```

> [@guillermohor](https://github.com/guillermohor) Yes I agree with that. There is a lot of work via UART. Soldering etc. and in my case 8 sensors. It is a pity that I changed the 0124 software which worked perfectly for me. I see that there is already version 0127 but it does not show up in the Zigbee2MQTT OTA

You don't need to solder anything. Just put dupont cables under the battery holder (one for GND and two for VCC). Second one connected to VCC and use to put pin P14 on high during activation. After activation hold data line connected to P14 to upload new firmware.

juha63, 2025-05-02:
@VivantSenior 

{"hwVersion":0} => B 1.4

hayvan96, 2025-05-05:
Hi, as for me, nothing has changed in my configuration, you can see I've upgraded from 0.1.2.4 to 0.1.2.6 beginning of April. Other sensors are starting to collapse as well, all are HW1. 4

[image]

hayvan96, 2025-05-05:
LQI : from 42 for the most remote to 132 for the closest (10 cm from the zigbee stick). 
Measurement interval : some have "unknown", other 10

RSSI : can't find the field, I'm using z2m

pvvx, 2025-05-05:
All my ZigbeeTlc devices have LQI over 200 (green lines). Their distances to routers are completely different. At 10 centimeters from the coordinator the LQI will be 255.

[image]

Only Tuya and Xiaomi devices have a lower LQI value.

pvvx, 2025-05-05:
The higher transmit power of the Zigbee network coordinator can overpower the routers' signal, and battery-powered devices may assume that a direct connection to the coordinator will provide a better route. However, these battery-powered devices will still transmit at their (low) level, and increasing the transmit power does not mean that the coordinator will be able to receive these signals better.
Similarly, another situation is when the device has a high transmitter power. Then the coordinator or router hears the device well, but the response does not reach the device.

pvvx, 2025-05-05:
Xiaomi LYWSD03MMC HW B1. 4, FW: ZigbeeTlc v0.1.2.7
[image]
Communication via the standalone [router](https://github.com/pvvx/ZigbeeTLc/issues/194#issuecomment-2848024812) specified earlier. Between the router and LYWSD03MMC: distance 5 meters, wooden wall, + distance 1.5 meters.

pvvx, 2025-05-05:
> I don't know if it has any consequence to link them to the coordinator instead of the nearest powered zigbee device.

In a ZigBee network, devices themselves choose a router based on the best signal. This is not always correct and their decision cannot be changed unless the device or router is moved.

hayvan96, 2025-05-05:
hmm, maybe I'll switch them back to BLE. I have not noticed any battery drain on other zigbee devices and nothing have changed in my network, for more than a year. Everything was okay with 0.1.2.3 for instance. But maybe all my Xiaomi thermometers being now under Zigbee are a bit too much for my environnement

kbullet, 2025-12-17:
Not sure if its gonna be of help for others, but for me downgrading to 0.1.2.2 fixed the LYWSD03MMC battery drain for me. On 0.1.2.6+ just kills the battery at around ~3months ish

[image]

pvvx, 2025-12-17:
> Not sure if its gonna be of help for others, but for me downgrading to 0.1.2.2 fixed the LYWSD03MMC battery drain for me. On 0.1.2.6+ just kills the battery at around ~3months ish

The hygrometer's power graph shows that your Zigbee network is constantly dropping.

kbullet, 2025-12-18:
> > Not sure if its gonna be of help for others, but for me downgrading to 0.1.2.2 fixed the LYWSD03MMC battery drain for me. On 0.1.2.6+ just kills the battery at around ~3months ish
> 
> The hygrometer's power graph shows that your Zigbee network is constantly dropping.
> 
> Newer versions now detect the network more aggressively after network loss. The Zigbee network is not designed to power down the coordinator and routers.

it does? Haven't noticed any network dropouts at all, mostly table full msgs with the 0.1.2.2. But everything else seems stable.

Maybe the HA graph had some funky rendering going on here's a more detailed ver

[image]

pvvx, 2025-12-18:
The normal graph has a smoother decline, without obvious dips.
The collapse and recovery indicate that there was very high consumption, but then it decreased.
This is possible if the hygrometer loses connection to the network for a long time, but the connection is restored after a while.

pvvx, 2025-12-18:
Average battery charge decline rates for LYWSD03MMC at +20°C to +25°C with default settings.

In the middle of the charge range (60% to 40%):
• For the cheapest CR2032 batteries - two months.
• For good CR2032 batteries - one month.
This means the charge decline rate on the graph is up to 10% per month for the cheapest CR2032 batteries.
A recent experiment with a $0.1 CR2016 battery on a test LYWSD03MMC B1.1:

[image]
(December 3 – the battery was disconnected and tests were performed with the new firmware version)

The battery charge level crosses the 50% level at a rate of decline of approximately 10-12% per month.

nilo85, 2026-01-06:
Just to add my two cents, I have previously been able to run the BLE firmware for year+, and now I recently changed all mine to run zigbee and they drain a CR2032 in weeks…