pvvx/ZigbeeTLc #195: Idea: user configurable transmission power

ss89, 2025-04-27:
I do have some thermostats using this firmware and i'm running into situations where they stop sending (or better said: i don't receive data) anymore after placing them into their spots.

Since this firmware has intentionally limited the transmit power to something low to save battery, i'd like to suggest a feature for making it less painful (no-solder work) for people with the same issue.

My suggestion is to keep the currently low transmit rate, but make it configurable via zigbee.
This should allow me to register a device near the router where it would work and gradually increase the transmit rate until i know that it is working stable.

I have no clue if this is easily doable, but i suppose that even if you can't update the transmit power on the fly, it might be just a mcu reboot away.

Let me know if there's any questions to my proposal.
It would be a super convenience feature!

pvvx, 2025-04-27:
CR2032 does not give out the required current to increase the transmission power.

pvvx, 2025-04-27:
> Since this firmware has intentionally limited the transmit power to something low to save battery, i'd like to suggest a feature for making it configurable at runtime (no-solder work).

No restrictions or reduced power are used. You invent something incomprehensible to anyone.

> My suggestion is to keep the currently low transmit rate as a default, but to make it configurable via Zigbee.
This should allow me to register a device near the router where it would work and gradually increase the transmit rate until i know that it is working stable in the destination spot (e.g. with 1 wall in between).

What other transmission speed? What is it? What are you talking about?

All transmission intervals in FW Zigbee TLC are tuned through the Zigbee 3.0 functions.

ss89, 2025-04-28:
> What other transmission speed? What is it? What are you talking about?

sorry, i was talking about transmit power, not rate.

The situation for me is as follows: I'm living in a house.
There are two Zigbee routers (Ikea bulbs, updated firmware) 2 + 5 Meters away from the thermometer (TZ3000).
There's a brick wall (well insulated, likely something that absorbs radio as i regularly have cell phone issues too right near the window) between the thermometer and the routers.
The thermometer looses connection every ~2 days all of a sudden. As soon as i bring it inside it immediately picks up and i receive new data.
I do in fact use different 2xAAA battery powered devices (with Li-Ion rechargeable cells).
As a matter of fact i also want to mention that i tested a new thermometer (ZYZTH02) with original firmware much further away outside and it reported just fine.
I am not willing to switch to bluetooth, because then i have one more wireless technology in here.

So in my point at least giving the option to change the transmit power (to a certain max) is a logical benefit for devices where you really need it, even considering all of your 5 points stating why you are against it.
I do see your point that it doesn't make sense for CR cells, but that is something i specifically avoided when picking a device for purchase.
Maybe it could be only added for devices that are known to have AAA batteries?

pvvx, 2025-04-28:
> Maybe it could be only added for devices that are known to have AAA batteries?

For devices with AAA batteries, the transmitter power is increased by default.