pvvx/ZigbeeTLc #197: How can I update ZigbeeTLc vesion over Z2M

davidsarkany, 2025-05-09:
Hello.

I have a pair [LKTMZL02](https://pvvx.github.io/LKTMZL02)
I installed ZigbeeTLc firmware with ts0201_tuya2z.json then I set the following OTA index file:
https://raw.githubusercontent.com/pvvx/ZigbeeTLc/refs/heads/master/bin/index_v0127.json

I added a custom external_converters for the OTA.
`import * as m from 'zigbee-herdsman-converters/lib/modernExtend';
```js

import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['LKTMZL02-z'],
    model: 'LKTMZL02-z',
    vendor: 'Tuya',
    description: 'Automatically generated definition',
    extend: [m.battery(), m.temperature(), m.humidity()],
    ota: true,
    meta: {},
};
```
I see the devices on the OTA page, but the LKTMZL02 with older firmware can't find any new updates.
[image]

Ctrl-F4, 2025-05-10:
Tested, seems to work. Your LKTMZL02 devices are already on v0.1.2.7 ? 

[image]

davidsarkany, 2025-05-11:
@Ctrl-F4 You are right. Firmware build date different, but file version same (19345409)

closed 2025-05-11 by davidsarkany