pvvx/ZigbeeTLc #198: CGG1 don`t work in Z2M

xmanslava, 2025-05-13:
Hello!
Yesterday I flashed this thermometer, an old model from 2021, to the ZigBee firmware. The home assistant does not support this device. I just want to change the sensor's reporting period, but it complains and won't let me change it.
Can you tell me where to get a converter for this device? 
And the second question I want to return the previous firmware of the bluetooth 5.2. How to switch back from ZigBee to your firmware.

[image]

I also noticed that the bluetooth version of the firmware works for a very long time and the temperature graph is very beautiful, picture for example:

[image]

igo-r, 2025-05-13:
@xmanslava Hello!
Zigbee2mqtt does not yet have a convector for this device.
I will add support for it today, it will be available with the next zigbee2mqtt update.
And I will send the External converter here so that you can use it now.

xmanslava, 2025-05-13:
Thank you for your prompt response, I'll be waiting!

igo-r, 2025-05-14:
https://github.com/Koenkk/zigbee2mqtt.io/pull/3781

igo-r, 2025-05-14:
https://github.com/Koenkk/zigbee-herdsman-converters/pull/9337

pvvx, 2025-12-18:
Instead of a smiley, a central stripe is used whenever possible.