pvvx/ZigbeeTLc #199: No periodic display refresh for MHO-C401N

igo-r, 2025-05-13:
Прежде всего хочу поблагодарить вас за проделанную работу. Ваша прошивка делает устройство еще лучше.

Я имею несколько устройств MHO-C401N.
Заметил как в течении не большого времени использования с вашей прошивкой на экране потерялось контрастность отображаемых сегментов, а также появились призрачные следы на сегментах экрана. Особенно это стало заметно в сравнении с новым устройством на оригинальной прошивке.

Здесь уже описывалась данная проблема для MHO-C401 (old): https://github.com/pvvx/ATC_MiThermometer/issues/43
И было исправлено внедрением алгоритма обновления сегментов.

Было бы замечательно иметь такой алгоритм для MHO-C401N, как в оригинальной прошивке.

Сейчас я делаю это с помощью автоматизации выключения и включения дисплея, чтобы очищать экран от призрачных следов.
Но это очень плохая практика, в связи с повышеным расходом энергии элемента питания на обработку zigbee команд.
А также после включения экрана, экран остается белым до следующего обновления данных, это от 10 секунд.

pvvx, 2025-05-14:
В MHO-C401N установлен дисплей не требующий refresh.
Старые типы E-Ink требуют обновления, а новые - нет.
В старых вариантах прошивок для MHO-C401N было включено обновление через тысячу выводов изображения. Но это ничего не давало.
В итоге проблема не с обновлением, а с управлением контроллера дисплея. 
Это проявляется, ели перепутаны прошивки MHO-C401 и MHO-C401N.

pvvx, 2025-05-14:
Дисплей и контроллер MHO-C401 и MHO-C401N отличаются по скорости обновления данных в 10 раз. 
Возможно что контроллер MHO-C401N делает refresh сегментов во время каждого вывода нового изображения, но этого процесса не видно из-за быстрого исполнения. По этому отдельный refsresh в MHO-C401N не применяется, а ошибки могут быть только в управлении контроллером этого дисплея.

igo-r, 2025-05-14:
> В MHO-C401N установлен дисплей не требующий refresh. Старые типы E-Ink требуют обновления, а новые - нет. В старых вариантах прошивок для MHO-C401N было включено обновление через тысячу выводов изображения. Но это ничего не давало. В итоге проблема не с обновлением, а с управлением контроллера дисплея. Это проявляется, ели перепутаны прошивки MHO-C401 и MHO-C401N. И несколько версий назад была бета версия прошивки где есть проблемы в работе с контроллером дисплея. Там проявлялись серые сегменты. Исправляется новой версией прошивки.

У меня модели MHO-C401N и установленная прошивка также MHO-C401N. На обоих стояли Ver 0.1.2.6, а теперь Ver 0.1.2.7.
Но проблема с сегментами осталась. Да они слабо заметны, но в сравнении с другим устройством на оригинальной прошивке это ощутимо.
Выключение и включение дисплея решает эту проблему. Пропадают остатки черного на пустых сегментах, а темные сегменты становятся более контрастнее.

pvvx, 2025-05-14:
> На оригинальной прошивке MHO-C401N обновление присутствует.

Я ни разу не видел, чтобы там происходило обновление, как это происходит у MHO-C401. 
На MHO-C401 этот процесс длится более секунды и там всё мигает. И такой refresh производится каждый 32-ой вывод, что всегда попадется на глаза.

pvvx, 2025-05-14:
Значит ещё какая-то версия с старым типом дисплея.

igo-r, 2025-05-14:
> Значит ещё какая-то версия с старым типом дисплея. MHO-C401 уже было более 3-х типов и производителей.

Производитель MiaoMiaoCe

Позже могу разобрать устройство и сфотографировать нужные элементы, если это может чем-то помочь.

pvvx, 2025-05-14:
Фотография ничего не решит.
Минимально нужен лог с логического анализатора всех выводов шлейфа к контроллеру E-ink во время "refresh".
Но лучше документация на этот контроллер.

igo-r, 2025-05-14:
Если бы документация была доступна в сети, думаю вы давно бы решили этот вопрос 😁

pvvx, 2025-05-14:
> Если бы документация была доступна в сети, думаю вы давно бы решили этот вопрос 😁

У некоторых, возможно, есть доступ туда, куда у других нет. 

Я уже много чего перепробовал, прежде чем вставить ту задержку, когда был вариант с проявлением серых сегментов на MHO-C401N в связи со сменой алгоритма и SDK в ZigbeeTlc. Ничего лучшего чем сейчас не вышло.

igo-r, 2025-05-14:
Печально, пока значит придется выключать и включать дисплей через автоматизацию в HА.
Так как за неделю использования одного устройства, проявление серых сегментов было крайне выраженым.

Сейчас с автоматизацией гораздо лучше, но излишние команды на устройство с таким слабым элементом питания, это плохо. Плюс очень часто после повторного включения экрана, экран долгое время пустой, что крайне режет глаз.

pvvx, 2025-05-14:
>  Плюс очень часто после повторного включения экрана, экран долгое время пустой, что крайне режет глаз.

Это значит у вас установлен большой интервал измерения. Экран обновляется когда есть какие либо изменения для отображения. А после вкл.выкл экрана отображение происходит после опроса датчика с установленным пользователем интервалом.

igo-r, 2025-05-14:
> > Плюс очень часто после повторного включения экрана, экран долгое время пустой, что крайне режет глаз.
> 
> Это значит у вас установлен большой интервал измерения. Экран обновляется когда есть какие либо изменения для отображения. А после вкл.выкл экрана отображение происходит после опроса датчика с установленным пользователем интервалом.
> 
> И ещё кто-то верит в рекламные сказки о том, что E-ink потребляет меньше чем типовой LCD... На обновление изображения у E-ink уходит уйма энергии и во всех MHO-C401(N) батарейки работают раза в два меньше, чем у термометров с LCD. E-Ink контролер меняет изображение (к примеру один сегмент) на MHO-C401 более 1.6 секунды. И всё это время у него работает повышающий напряжение преобразователь... У MHO-C401N процесс быстрее, но во столько же раз и ток потребления при отображении больше... А в CGG1 батарейка работает дольше за счет того, что она в 4-ре раза больше, если учитывать сколько может отдать батарейка CR2032 и CR2450 для текущих использованных чипов с их требуемым пиковым потреблением в 8 мА :) Термометры с CR2450 и большим LCD работают от батареи не менее 1.5 года, а типично более 2+ года.

10 секунд, по умолчанию, бывает в течении 10 и обновляет, но изредка бывают задержки до минуты.

igo-r, 2025-05-14:
> И ещё кто-то верит в рекламные сказки о том, что E-ink потребляет меньше чем типовой LCD... На обновление изображения у E-ink уходит уйма энергии и во всех MHO-C401(N) батарейки работают раза в два [...]

Потребление будет меньше только при условии долгого отображения статичных данных.
Но это явно не про температурный датчик, где значения меняются часто.
В моем случае, мне просто нравиться эта модель, серый цвет корпуса смотрится гораздо лучше.
Если рассматривать внешний вид E-ink тоже смотрится лучше LCD.

pvvx, 2025-05-14:
> Если рассматривать внешний вид E-ink тоже смотрится лучше чем LCD.

Это смотря какой. Худший дисплей имеет Xiaomi LYWSD03MMC. Именно по этому его чаше всех покупают.
Из Xiaomi лучший - MJWSD05MMC,  а MJWSD06MMC не совсем.

* v0.1.2.8 - Для MHO-C401N добавлено обновление E-Ink после 1024 обновлений дисплея.

igo-r, 2025-05-14:
> > Если рассматривать внешний вид E-ink тоже смотрится лучше чем LCD.
> 
> Это смотря какой. Худший дисплей имеет Xiaomi LYWSD03MMC. Именно по этому его чаше всех покупают :P У новых Tuya, аналогичных LYWSD03MMC, LCD контрастнее в несколько раз. И LYWSD03MMC дороже раза в три... Из Xiaomi лучший - MJWSD05MMC, а MJWSD06MMC не совсем.
> 
> MHO-C401(N) имеет мелкую батарею и для Zigbee вообще не годится - слишком малый срок работы CR2032 в нем.
> 
> * v0.1.2.8 - Для MHO-C401N добавлено обновление E-Ink после 1024 обновлений дисплея.

Согласен касаемо LYWSD03MMC, экран очень плохой.
MJWSD06MMC мне больше всего приглянулся из всех, но экран в живую не видел.

В MHO-C401 действительно очень слабый источник питания, но зато имеет тонкий стильный корпус. Я потестирую на сколько хватит штатных CR2032, но предполагаю что они не лучшего качества.
Скорее всего поставлю Duracell и увеличу частоту обновления до 30 секунд. Возможно еще подберу интервал отчетов оптимальный для себя, но не сильно частый. Это единственный выход с данным устройством. И нужно позаботится о максимально хорошем сигнале zigbee сети.

Спасибо за выпуск новой версии для MHO-C401N!

igo-r, 2025-05-15:
@pvvx Сделал пару фотографий, с двух устройств, возможно они смогут натолкнуть на мысль.

Словил на одном устройстве момент обновления дисплея добавленного вами в v0.1.2.8
Присмотрелся и заметил большое количество темных точек по всему дисплею, также сразу после выключения дисплея, имеется отпечаток последних данных.

[image]

Сделал фотографию также экрана второго устройства с прошивкой v0.1.2.8, где можно увидеть проблемные темные линии вокруг сегментов.

[image]

Такое ощущение что при обновлении на сегменты поступает подача меньшего напряжения чем положено либо на меньшее время чем нужно, для полного обновления сегмента.

igo-r, 2025-05-15:
На фотографиях выше я думаю отчетливо видно проблемы из за которых я поднял данную тему.

pvvx, 2025-05-15:
> На фотографиях выше я думаю отчетливо видно проблемы из за которых я поднял данную тему.

> Фотография ничего не решит.
> Минимально нужен лог с логического анализатора всех выводов шлейфа к контроллеру E-ink во время "refresh".
> Но лучше документация на этот контроллер.

igo-r, 2025-05-15:
> > Чего нет того нет. Если у вас имеется анализатор, давайте я поспособствую приобретению MHO-C401N данной сборки, в качестве своего вклада. Тем более они есть на озон.
> 
> У меня несколько работающих MHO-C401N. Именно по этому и писал вам, что в оф.версии не смог встретить событий обновлений экрана. А [логический анализатор](https://aliexpress.ru/item/1005001621950241.html) на али стоит в несколько раз меньше MHO-C401N. Теперь считаем, что он у вас есть.

В мои места с Али я буду ждать месяц-больше.(

pvvx, 2025-05-15:
> В мои места с Али я буду ждать месяц-больше.(

Т.е. нет паяльника? В фикс-прайс он стоит 1 единицу.
Может ещё у вас нет отвертки, чтобы открутить  крышку MHO-C401N?

У меня нет главного - события обновления на оф. прошивке и выдумать как это сделать я не умею.

igo-r, 2025-05-15:
> > В мои места с Али я буду ждать месяц-больше.(
> 
> Т.е. нет паяльника? В фикс-прайс он стоит 1 единицу. Может ещё у вас нет отвертки, чтобы открутить крышку MHO-C401N?
> 
> У меня нет главного - события обновления на оф. прошивке и выдумать как это сделать я не умею.

Именно поэтому я и предлагал приобрести устройство данной сборки для вас, чтобы вы смогли увидеть работу именно этого дисплея, либо возможно просто обновлений прошивки.

pvvx, 2025-05-15:
> Именно поэтому я и предлагал приобрести устройство данной сборки для вас, чтобы вы смогли увидеть работу именно этого дисплея, либо возможно просто обновлений прошивки.

Извините, но мне уже давно некуда девать все эти разнообразные термометры... Валяются заполненные коробки... Скоро начну выбрасывать в помойку, т.к. знакомым не имеющих всяких умных домов они не нужны, а тем кому были нужны уже выданы.
А времени на ковыряние в них уже почти нет в этом году, т.к. летом я отдыхаю от всяких работ на компьютерах и т.д...
И если хотите помочь - то предлагаю уточнить, что поменять в коде и опубликовать в [pull request](https://github.com/pvvx/ZigbeeTLc/pulls).

pvvx, 2025-05-15:
[XMWSDJ04MMC](https://pvvx.github.io/XMWSDJ04MMC/) на оф.прошивке. Альтернативного FW на него нет:
[image]

pvvx, 2025-05-15:
Один из первых CGG1-M на TLSR825x. Наверно года 4-ре работы:

[image]

igo-r, 2025-05-15:
> Один из MHO-C401N:
Ваш выглядит поживее, а еще с учетом сколько ему времени.
MHO-C401 вообще выглядит по дисплею само лучше.

> [XMWSDJ04MMC](https://pvvx.github.io/XMWSDJ04MMC/) на оф.прошивке. Альтернативного FW на него нет:
А вот здесь уже такие же аномали что и у моих MHO-C401N

pvvx, 2025-05-15:
> А вот здесь уже такие же аномали что и у моих MHO-C401N

Дык он старый уже. Деградировал.

pvvx, 2025-05-15:
[image]

[image]

igo-r, 2025-05-15:
> > А вот здесь уже такие же аномали что и у моих MHO-C401N
> 
> Дык он старый уже. Деградировал.

Вот в этом и проблема, что мои новые одному и месяца нет, а состояние будто лет 5 😁

Похоже проще было выбрать что-то с LCD дисплеем И мощным элементом питания.
Но по дизайну конечно приглянулась эта модель.
Еще устал от белых термотреов. Этот хорошо вписался на темной полке и сером холодильнике)

igo-r, 2025-05-15:
А на ваших MHO-C401N если выключать дисплей остается отпечаток последних показаний?
Интересно, это проблема всех MHO-C401N или в новых ревизиях действительно сменили дисплей, и поэтому добавили обновление в офф прошивку.

pvvx, 2025-05-15:
Покопался в архиве и нашел фото первого включения [LYWSD02](https://pvvx.github.io/LYWSD02/)

[image]

Сразу были видны засветы сегментов. Контрастность была немного больше, но не занчительно.
И это первое поколение E-Ink по классификации [викопедии](https://ru.wikipedia.org/wiki/%D0%AD%D0%BB%D0%B5%D0%BA%D1%82%D1%80%D0%BE%D0%BD%D0%BD%D0%B0%D1%8F_%D0%B1%D1%83%D0%BC%D0%B0%D0%B3%D0%B0). Аналогично и у MHO-C401, первых CGG1-M.
А у MHO-C401N - это второе поколение...

pvvx, 2025-05-15:
> А на ваших MHO-C401N если выключать дисплей остается отпечаток последних показаний? Интересно, это проблема всех MHO-C401N или в новых ревизиях действительно сменили дисплей, и поэтому добавили обновление в офф прошивку.

Его выпускают разные производители. Могут быть и подделки.
Сегменты, которые чаше меняются всегда оставляют след на всех типах и деградируют быстрее...

pvvx, 2025-05-15:
И в сообщении "Один из MHO-C401N" никакого "обновления" (refresh) в прошивке нет и не используется. Так и работает...

igo-r, 2025-05-15:
> Но производства Azarton так и не выловился... И больше не хочу с ними возиться, заказывать ещё и т.д.

Судя коробке у меня именно MiaoMiaoCe
Уже сам начинаю задумываться а стоит ли возиться с этими MHO-C401N с учетом установленного маленько элемента питания активно использовать его не получится, все будет строится вокруг компромиссов.

Возможно действительно стоит приглядеться к датчикам с LCD дисплеем и мощным элементом питания что в zigbee сети явно не помешает.
Что можете посоветовать?

[MJWSD06MMC](https://pvvx.github.io/MJWSD06MMC) как в целом? И как по стабильности zigbee прошивка под его?

igo-r, 2025-05-15:
И еще раз выражаю вам благодарность за вашу большую проделанную работу по альтернативным прошивкам на такой большой список устройств.

igo-r, 2025-05-16:
@pvvx 
> У меня нет главного - события обновления на оф. прошивке и выдумать как это сделать я не умею.

Хочу уточнить.
Я только что обновил одно из устройств MHO-C401N на оф. прошивку [v1.0.0_0016](https://pvvx.github.io/MHO_C401N/bin/ff-mho-c401-1.0.0_0016.zip)
Там также имеется обновление дисплея, притом частое, я специально создал условия для частой смены данных, и на каждом 32 смене данных происходит обновление дисплея.

igo-r, 2025-05-16:
https://github.com/user-attachments/assets/dcc1b131-aeb8-4f48-a41a-f27096cf15c7

Записал видео момента обновления дисплея при 240 кадрах в секунду.