pvvx/ZigbeeTLc #201: New TS0201 variant: TS0201_TZ3000_rdhukkmi

vulpes2, 2025-05-22:
Looks like there's another variant of the TS0201, `_TZ3000_rdhukkmi`. Sensor is an SHT30 and everything is wired differently compared to the two supported TS0201 models. There is no silkscreen text on either side of the PCB.

|ZTU|Board|Notes|
|-|-|-|
|B4|LED|active low|
|C0|Switch|active low, 10k pull-up|
|D7|SHT30 ALERT|10k pull-up|
|C2|SHT30 SDA|10k pull-up|
|C3|SHT30 SCL|10k pull-up|
|-|SHT30 ADDR|connected to GND|
|-|SHT30 nRESET|10k pull-up|

[image]

[image]

pvvx, 2025-05-22:
Update v0.1.2.8: For TS0201 enabled sensors CHT8305, AHT20..30, SHT4X, SHT30.
In the BLE version, all types of sensors except SHTC3 have long been included.

vulpes2, 2025-05-23:
Tested the new build and it works great, thanks!

closed 2025-05-23 by vulpes2

Korenchkin, 2025-08-28:
so,today(more like yesterday already) i got this sensor(i always buy more so i can play),so one dead already :D ,i flashed (i think) 1141-d3a3-01993001-TH02Zz_v0129.zigbee (after i discovered,how to efficiently work with override json) and that only 3 or 2 things matter - manufacturer id and image type (and version)...and i think nobody really tells you ota mechanics read those versions from file and tells you nothing if it is not right...

i got this sensor somehow working,but as a different device(021c) and button was dead,led was turned on(i didn't expect this to work at all,so i was surprised),now with all that power,i went crazy,selected image type 225 (that should be this device,right?),tried to force it,no go...yes you need new image type,so i used this image(0225) as a base,hexedited header to 021c (that was on after first wrong flash),and still didn't want to eat it,mqtt downgrade did not help,so again back to hexedit and change version(probably minor number) to +1 and it went to 100%...and that was it,so i think i'm gonna try serial way,once i find swo(swd?) and reset...this is just a story,if anyone has some advice....i need some sleep now
flashaholics anonymous send regards