pvvx/ZigbeeTLc #205: Not able to flash LYWSD03MMC B1.5 to Zigbee

MisterTB84, 2025-06-13:
Hi today I received my 6th and 7th LYWSD03MMC device, I'm very satisfied with the other 5.
I noticed when ordering it is a 2nd generation, don't know if that makes a difference?

Just like the other five I flashed it to the custom firmware, that worked fine.
Then I flashed to Zigbee: Update done after 110.658 seconds
Unfortunately the device still is in Bluetooth mode, looks like no flash has occured.

I tried both devices multiple times from different computers, alle the same result.
Any ideas?

21:27:33: Searching for devices
21:27:43: Get Advertising MAC (Web Experimental Features!)
21:27:43: TypeError: device.watchAdvertisements is not a function
21:27:43: Reconnect 1 from 5
21:27:50: Hardware Revision String: B1.5
21:27:50: Software Revision String: V5.2
21:27:50: Firmware Revision String: github.com/pvvx
**21:27:50: Detected custom Firmware
21:27:50: Hardware Version: LYWSD03MMC B1.5, Software Version: 5.2**
21:27:50: Custom config HEX string: 5507100a062804a931310ab4
**21:27:59: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01283001-Z03MMC.zigbee'...**
21:27:59: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01283001-Z03MMC.zigbee
21:27:59: Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01283001
21:27:59: APP Release: 0.1 Build: 2.8, Stack Version: 3.0 Release: 0.1
21:27:59: File size: 131364 bytes
21:27:59: Count: 8211
21:28:03: Start DFU
21:28:04: Clear Ext.OTA region (129kib)...
21:28:06: Waiting Start ext.OTA...
21:28:13: Clean Flash OTA region: 0x040000, size: 132 kbytes. Go OTA...
**21:30:04: Update done after 110.658 seconds**
21:31:33: Reconnect to: ATC_2B9C23
21:31:42: Hardware Revision String: B1.5
21:31:42: Software Revision String: V5.2
21:31:42: Firmware Revision String: github.com/pvvx
21:31:42: Detected custom Firmware
21:31:42: Hardware Version: LYWSD03MMC B1.5, Software Version: 5.2
21:31:42: Custom config HEX string: 5507100a062804a931310ab4

darozmini, 2025-06-15:
I flashed LYWSD03MMC with 1141-020a-01283001-Z03MMC.zigbee and can't connect to aqara hub m3 nor bluetooth. I would be really grateful if anybody could give me a clue what went wrong
20:41:47: Get Advertising MAC (Web Experimental Features!)
20:41:47: Connecting to: LYWSD03MMC
20:41:49: MiVer5, DevID: 0x055B-LYWSD03MMC, FnCnt: 19, CtrID: 0x5858 Not bound, Standard certification, MAC: A4C13841DEF7, Data encrypted
20:41:58: Hardware Revision String: B1.4
20:41:58: Software Revision String: 0159
20:41:58: Firmware Revision String: 2.1.1_0159
20:41:58: Detected Mi device
20:41:58: Connected
20:43:48: Send Login, please wait...
20:43:48: Received device infos are correct
20:43:49: Login successful
20:45:28: Load firmware file 'bin/ATC_v52.bin'...
20:45:28: File: bin/ATC_v52.bin
20:45:28: File size: 86324 bytes
20:45:28: Add signature file.
20:45:28: Count: 5718
20:45:31: Start DFU
20:46:15: Update done after 43.453 seconds
20:46:21: Disconnected.
20:46:30: Reconnect to: LYWSD03MMC
20:46:32: Hardware Revision String: B1.4
20:46:32: Software Revision String: V5.2
20:46:32: Firmware Revision String: github.com/pvvx
20:46:32: Detected custom Firmware
20:46:32: Hardware Version: LYWSD03MMC B1.4, Software Version: 5.2
20:46:32: Custom config HEX string: 5507100a062804a9313100b4
20:53:49: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01283001-Z03MMC.zigbee'...
20:53:49: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020a-01283001-Z03MMC.zigbee
20:53:49: Zigbee OTA Info: ManCode: 1141, ImageType: 020A, FwVer: 01283001
20:53:49: APP Release: 0.1 Build: 2.8, Stack Version: 3.0 Release: 0.1
20:53:49: File size: 131364 bytes
20:53:49: Count: 8211
20:53:53: Start DFU
20:53:55: Clear Ext.OTA region (129kib)...
20:53:57: Waiting Start ext.OTA...
20:54:03: Clean Flash OTA region: 0x040000, size: 132 kbytes. Go OTA...
20:54:47: Update done after 43.362 seconds
20:54:51: Disconnected.
20:54:57: Reconnect to: LYWSD03MMC
20:55:05: NetworkError: Connection Error: Connection attempt failed.

isinva, 2025-06-15:
It won't connect via Bluetooth anymore, it's now zigbee )) Try opening the lid, short-circuiting reset and gnd for more than 15 seconds (good contact is needed, clean the pads thoroughly, it's much harder to short-circuit than it seems). Something like O O O should appear on the screen. It should connect after that.

pvvx, 2025-06-15:
> Once it displayed even the bluetooth icon blinking for a few seconds.

This means that it has connected to the Zigbee network. It has been registered by some Zigbee network.
Otherwise, the BLE icon will be constantly displayed indicating that no network was found.

> When I short-circuit reset and gnd temperature changes from Celsius degrees to Fahrenheit. It blinked a bluetooth icon for a few seconds.

This means that you did not hold the contacts for more than 15 seconds for a full reset. After 15..17 seconds, the current temperature will be displayed in Celsius format.

darozmini, 2025-06-15:
OK, this time I made a full reset 15s and Celsius degrees displayed. Do you know by any chance how to connect it now to aqara hub m3?

pvvx, 2025-06-15:
On Xiaomi hub-s, typical Zigbee 3.0 does not work. Devices are registered, but readings are not received. Xiaomi has its own protocol for Zigbee devices.

pvvx, 2025-06-15:
I don't have "aqara m3". I have other Xiaomi and Tuya hubs. On most of them, none of the standard Zigbee 3.0 works. Tuya needs Tuya Zigbee, Xiaomi needs Xiaomi Zigbee, which work according to their standard.
If you reprogram the HUB into a Zigbee coordinator with alternative firmware, everything works.
Many other hubs that support Zigbee 3.0 also work.

darozmini, 2025-06-15:
So, how can I connect this sensor to aqara app or mi app after flashing to zigbee? 

> I don't have "aqara m3". I have other Xiaomi and Tuya hubs. On most of them, none of the standard Zigbee 3.0 works. Tuya needs Tuya Zigbee, Xiaomi needs Xiaomi Zigbee, which work according to their standard. If you reprogram the HUB into a Zigbee coordinator with alternative firmware, everything works. Many other hubs that support Zigbee 3.0 also work. For example, Yandex stations, etc. But it is impossible to update the firmware on them.

pvvx, 2025-06-15:
> So, how can I connect this sensor to aqara app or mi app or google home after flashing to zigbee ?

None of this will work unless you reprogram the hub.

darozmini, 2025-06-15:
IWhat should I do to connect it to Home Kit? Should I buy a usb dongle as a zigbee hub?

> None of this will work unless you reprogram the hub.

pvvx, 2025-06-15:
The manufacturer has limited you so that you buy only its devices.
The use of alternative firmware and other things is prohibited in the user agreement in Mi-Home, which is displayed when connecting any Xiaomi device.
Similarly, it is prohibited to use Xiaomi servers for your own needs. Only for Xiaomi devices. Otherwise, you violate the law.

darozmini, 2025-06-15:
Ok, now I get it. Thank you so much. Is there a simple way to go back to ble version if I flashed it with 1141-020a-01283001-Z03MMC.zigbee?

darozmini, 2025-06-15:
I would like to buy Zigbee gateway with Zigbee OTA capability. Can you recommend any?

pvvx, 2025-06-15:
Xiaomi Gateway 3 and some other hubs can be connected to work with standard Zigbee:
https://github.com/AlexxIT/XiaomiGateway3

Thermometer programs are tested only for work with Home Assistant. Mainly with built-in "ZHA" integration. Z2M has a separate installer and is not much considered, as it is more difficult for users.

pvvx, 2025-06-15:
For many SMART Home programs, it is best to use a USB key that can always be reprogrammed to support various formats, including new ones - Matter.

https://darkxst.github.io/silabs-firmware-builder/

The outdated USB-Zigbee is better not to buy ... They have a weak level of receiving signal and restrictions on the number of connected devices.

closed 2025-06-24 by MisterTB84