pvvx/ZigbeeTLc #210: Missing capacitor

codewise-nicolas, 2025-06-24:
Ive been reading a few issues about battery issues/life with the FW.
The topic of a missing capacitor keeps coming up.
https://github.com/pvvx/ZigbeeTLc/issues/8
https://github.com/pvvx/ZigbeeTLc/issues/37#issuecomment-2831314917
`Installing the missing 22..47 μF capacitor (tantalum or ceramic) in the LYWSD03MMC in the power supply increases the operating time of CR2032 by 40%.`

I haven't been able to find any information however on where this cap should go or what type.
Can you kindly let me know more information on this or a link to somewhere I can find more info.

Thanks

pvvx, 2025-06-27:
The voltage of more than 3.6 V immediately leads to the degradation of the TLSR825X chips.
Such a chip will have an increase in nutrition consumption forever. Not repaired. Only replacing the chip.

pvvx, 2025-06-27:
If you are interested in methods to achieve maximum battery life from CR2032, read this:
https://www.ganssle.com/reports/ultra-low-power-design.html
For the operating conditions of BLE thermometers, it is necessary to take into account not 10 years of operation, but at least 1 year. Then the capacitor will help.

FireFury0, 2025-06-30:
> The voltage of more than 3.6 V immediately leads to the degradation of the TLSR825X chips. Such a chip will have an increase in nutrition consumption forever. Not repaired. Only replacing the chip.
> 
> Tested on 5 different devices with TLSR825X. As well as a dozen TLSR825X with random errors when working with a power supply of more than 3.6V. None of them could withstand it and all of them degraded and began to consume increased power current, but they remained operational.
> 
> [@FireFury0](https://github.com/FireFury0) - Due to these chip characteristics, your experiment is no longer of interest to anyone.

Mitigated with a 1N4148 diode between the battery positive contact and the rest of the circuit.

[image]

Method: cut the track between the battery terminal and the pad marked P2, use a multimeter to make sure its really cut because it's a tiny track with a black conformative coating, solder a diode between the battery terminal and P2, with the cathode connected to P2.  Easily reversible, in fact there is enough space inside the case to add a small switch to bridge the diode out if you want the option of easily going back to a primary cell.  T5 Torx bit required to open the case (2 screws to hold the board in contact with the LCD's zebra strip)

The voltage is reading as 3500mv with a fully charged (4200mv) LIR2032, so has the expected 700mv drop across the diode.  The minimum voltage for TLSR825X is listed as 1.8v, so would equal a battery voltage of 2500mv - far below the 3000mv safe minimum of a Li-Ion cell.  A Schottky diode could probably also be added in series, to add a further voltage drop if necessary.

elmystico, 2025-07-01:
hey I've just recently put some ceramic samsung caps on C24 pads (1206; 100uF; 6,3V; X5R; (SMD); CL31A107MQHNNNE; Samsung)
so yes this is the right place.
I was doing this with soldering station with hot air. I think it's doable with precise small soldering iron as well.
Looking at battery drainage on history graphs well it works but after a year I can tell if it was worth it :-)

elmystico, 2025-07-01:
[image]

 Actual job (my wife was soldering :-) )

pvvx, 2025-07-01:
CL31A107MQHNNNE has a current leakage comparable to the consumption of a thermometer in sleep mode (between transmissions and indication changes).
It is advisable to use capacitors for a significantly higher voltage - from 25 V and optimal capacity. The higher the operating voltage and the smaller the capacity - the less leakage in such capacitors.

pvvx, 2025-07-01:
DeepSeek: What is the leakage current of a typical 1206 X5R 100µF 6.3V ceramic SMD capacitor?
...
**Typical Expected Range:**
_For a real 100µF, 6.3V, X5R capacitor in a 1206 package, expect a leakage current in the range of 5–20 µA (microamperes) at room temperature (25°C)._
_A value of 10–15 µA is a very realistic estimate._
_This is 3-4 orders of magnitude (1000-10,000 times) higher than a typical 1206 ceramic capacitor with 1µF or 10µF capacitance!_
...

Typical average consumption of Xiaomi LYWSD03MMC B1.4 in BLE with default settings = **14 μA**.

pvvx, 2025-07-01:
A suitable 100 μF capacitor will cost as much as a thermometer.
Therefore, it is necessary to choose optimally for all parameters. Not for maximum capacity and not for low voltage.
But with Zigbee, this will still give a small effect.
An example that Qingping used in https://pvvx.github.io/CGDK2/.
[image]
But CR2430 is used there...

pvvx, 2025-07-02:
> Am I understanding that the reason for not using the capacitor is that the leakage current would wipe out any gains from a longer life of the cr2032? and the more expensive caps are not worth it given the low cost of cr2032's.
> 
> However if the goal is ~1-2 year on a cr2032 then the cap will be of help in extending life is my understanding.

If we take into account 1..2 years of operation, then the capacitor will save the CR2032 element from degradation, which is ~40% when working without a capacitor (in BLE!).
Zigbee has longer transmission and reception times. As a result, a larger capacity will be required. But a larger capacity has a higher leakage current... Therefore, it is necessary to choose a compromise. And what kind of compromise can only be found experimentally on hundreds of devices with observations over several years.

pvvx, 2025-08-12:
@FireFury0 - Everything is completely different from what you think.

> if the device moves out of range of the zigbee mesh, it drains the battery fast. I know that @pvvx has said that "zigbee does not like coordinator outages", but no other battery operated zigbee devices on my network drain the battery like that when they can't see the network. It would be nice if the device could drop into a low power mode when the network vanishes and leave long intervals before attempting to reconnect.

When the network is lost, pauses from 9 to 240 seconds are made to poll for network restoration. Immediately after the connection is lost, the first 7 pauses are 9 seconds, then 46 pauses of 55 seconds, then the pause increases by 1 second up to 240 seconds (55..240).
When a network is detected, a “rejoin” is required. But this still consumes more than normal operation. In the standard and in the basic SDK, there are no pauses at all.
Do not compare with the Tuya or Xiaomi protocol. They do not have Zigbee 3.0, but their own proprietary protocol.

> there is no protection to stop the battery being discharged too much. You can monitor it through Home Assistant and manually disconnect the cell once the voltage drops too low, but it would be nice if there was a way to program the device to shut down when the voltage drops below a predefined level.

The firmware implements:
When the power supply reaches 2.0V, it switches to low power mode. The main chip consumes less than 0.2 μA. The remaining components - depending on which are used in a specific version of the thermometer. Components with shutdown functions are also switched to minimum power mode.

After 180 seconds of minimum power (chip and component sleep), a restart is performed with a check that the voltage is more than 2.2V. If the voltage is less, then the switch to low power mode is performed again for 180 seconds.

pvvx, 2025-08-12:
> I just want to report back on my experiment with running these on an LIR2032 cell. I've got 12 LYWSD03MMC sensors running the Zigbee firmware. 10 of them have a 1N4148 diode installed to drop the cell voltage and are powered by LIR2032 cells. The remaining 2 are unmodified and powered by CR2032 cells.

[image]

[image]

pvvx, 2025-08-12:
> I'll accept that they may not be fully compliant with the Zigbee standards, but they do work fine on my Zigbee network, and do not suffer from the same battery drain problem. So possibly some _optional_ non-compliance mode might be beneficial. I don't accept that standards compliance must be maintained at all costs in a situation where there are substantial benefits and little detriment from bending the standards slightly (which is presumably why Tuya / Xiaomi have done exactly this).

This requires support from the network coordinator. That is, special software is needed on the coordinator's side.

> This statement confuses me: devices that have direct bindings (e.g. light switches that are bound directly to light bulbs, or control a group of light bulbs) work fine without the coordinator. Obviously anything that relies on HA automations is going to fail, but it seems odd to say that this is a "very low level of reliability and cannot be used to control home devices"

When you turn off the router, your entire network segment goes down. And no notifications or events will reach the coordinator...

> It would be quite useful if that were configurable - I would like to be able to shut down when the voltage reached 2.3v (which equates to a battery voltage of 3v on my modified devices).

To do this, you can compile the firmware from the source codes with any changes or additions.