pvvx/ZigbeeTLc #215: LYWSD03MMC - Z2M no update since OTA update

sy6sy2, 2025-07-02:
Yesterday I update one of my LYWSD03MMC device from 0.1.23-3.0.01 to 0.1.28-3.0.01 directly from Zigbee2MQTT.

The update process worked fine and the device screen is working correctly (no BT or battery indicator).
But since the update of the device I do not receive any sensor update on Zigbee2MQTT and the interview failed each time.

I tried to restart Zigbee2MQTT and unplug/plug the battery of the device without success.

My last hope is to reset the device with the reset pin but before that I opened this issue 😉 

Thanks!

juha63, 2025-07-02:
Did you read the README?????

_Warning(!): Firmware version 0.1.2.6 is built on a new version of SDK from Telink. **Requires reset of all previous settings and new binding**. Old values from previous versions remembered by the device are incompatible._

sy6sy2, 2025-07-02:
Yes I read it. But I guess I misunderstand the message, I thought that the device was able to reset its settings and rebind itself to the zigged network ...
After a manual reset and re bind everything works as expected!
Thank you!!