pvvx/ZigbeeTLc #222: Cannot flash Zigbee firmware to MJWSD06MMC

dariuszwisniewski, 2025-07-15:
I have successfully flashed my MJWSD06MMC with a custom firmware: MJ6_v53.bin via https://github.com/pvvx/ATC_MiThermometer.
Unfortunately, I cannot flash it with the Zigbee firmware (1141-020d-01293001-Z06MMC.zigbee). It just stuck on one of the first steps, and nothing happens. I have already tried on MacBook, Windows PC and Android tablet, and the result is the same. 

Am I doing something wrong, or is it a bug? 

```
10:31:28 AM: Searching for devices
10:31:38 AM: Connecting to: MJ6_2E0B57
10:31:43 AM: Hardware Revision String: V1.3
10:31:43 AM: Software Revision String: V5.3
10:31:43 AM: Firmware Revision String: github.com/pvvx
10:31:43 AM: Detected custom Firmware
10:31:43 AM: Hardware Version: MJWSD06MMC V1.3, Software Version: 5.3
10:31:43 AM: Custom config HEX string: 5587100a065004a931310d5a
10:31:48 AM: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020d-01293001-Z06MMC.zigbee'...
10:31:48 AM: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020d-01293001-Z06MMC.zigbee
10:31:48 AM: Zigbee OTA Info: ManCode: 1141, ImageType: 020D, FwVer: 01293001
10:31:48 AM: APP Release: 0.1 Build: 2.9, Stack Version: 3.0 Release: 0.1
10:31:48 AM: File size: 130052 bytes
10:31:48 AM: Count: 8129
10:31:50 AM: Start DFU
10:31:52 AM: Clear Ext.OTA region (128kib)...
10:31:52 AM: Information: Current OTA region: 0x000000, size: 128 kbytes
```

pvvx, 2025-07-15:
Try updating again to MJ6_v53.bin via BLE OTA. And after that, Zigbee.

dariuszwisniewski, 2025-07-15:
> Try updating again to MJ6_v53.bin via BLE OTA. And after that, Zigbee.

I have already tried that multiple times, this is the most recent log:
```
11:29:46 AM: Searching for devices
11:29:51 AM: Connecting to: MJ6_2E0B57
11:29:55 AM: Hardware Revision String: V1.3
11:29:55 AM: Software Revision String: V5.3
11:29:55 AM: Firmware Revision String: github.com/pvvx
11:29:55 AM: Detected custom Firmware
11:29:55 AM: Hardware Version: MJWSD06MMC V1.3, Software Version: 5.3
11:29:55 AM: Custom config HEX string: 5587100a065004a931310d5a
11:30:05 AM: Load firmware file 'bin/MJ6_v53.bin'...
11:30:05 AM: File: bin/MJ6_v53.bin
11:30:05 AM: File size: 81492 bytes
11:30:05 AM: Count: 5094
11:30:08 AM: Start DFU
11:31:07 AM: Update done after 57.944 seconds
11:31:08 AM: Disconnected.
```

Then I reconnect and try with the Zigbee firmware:

```
11:31:23 AM: Searching for devices
11:31:25 AM: NotFoundError: User cancelled the requestDevice() chooser.
11:31:25 AM: Reconnect 1 from 5
11:31:45 AM: Searching for devices
11:31:50 AM: Hardware Revision String: V1.3
11:31:50 AM: Software Revision String: V5.3
11:31:50 AM: Firmware Revision String: github.com/pvvx
11:31:50 AM: Detected custom Firmware
11:31:50 AM: Hardware Version: MJWSD06MMC V1.3, Software Version: 5.3
11:31:50 AM: Custom config HEX string: 5587100a065004a931310d5a
11:31:52 AM: NotFoundError: User cancelled the requestDevice() chooser.
11:31:52 AM: Reconnect 1 from 5
11:31:53 AM: Hardware Revision String: V1.3
11:31:53 AM: Software Revision String: V5.3
11:31:53 AM: Firmware Revision String: github.com/pvvx
11:31:53 AM: Detected custom Firmware
11:31:53 AM: Hardware Version: MJWSD06MMC V1.3, Software Version: 5.3
11:31:53 AM: Custom config HEX string: 5587100a065004a931310d5a
11:31:53 AM: Hardware Version: MJWSD06MMC V1.3, Software Version: 5.3
11:31:53 AM: Custom config HEX string: 5587100a065004a931310d5a
11:31:55 AM: Load firmware file 'https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020d-01293001-Z06MMC.zigbee'...
11:31:55 AM: File: https://raw.githubusercontent.com/pvvx/ZigbeeTLc/master/bin/1141-020d-01293001-Z06MMC.zigbee
11:31:55 AM: Zigbee OTA Info: ManCode: 1141, ImageType: 020D, FwVer: 01293001
11:31:55 AM: APP Release: 0.1 Build: 2.9, Stack Version: 3.0 Release: 0.1
11:31:55 AM: File size: 130052 bytes
11:31:55 AM: Count: 8129
11:31:57 AM: Start DFU
11:31:58 AM: Clear Ext.OTA region (128kib)...
11:31:59 AM: Information: Current OTA region: 0x020000, size: 128 kbytes
11:31:59 AM: Information: Current OTA region: 0x020000, size: 128 kbytes
```

And nothing else happens.

pvvx, 2025-07-15:
Fixed in TelinkMiFlasher v11.3