pvvx/ZigbeeTLc #229: several devices offline - they don't hear coordinator?

Korenchkin, 2025-09-06:
hello,
this is more like a question about zigbee network with your devices...

so,i have 4 devices now,one round display and 3x ts0201-like with your latest firmware (129,because it supports i2c thermometer inside),i have it really close ,about 2-3 meters from coordinator - zb-gw04 1.2 (firmware latest 7.4.x and now 8.2.1 for test),but given enough time they go offline from worst signal to best (in a matter of days)
i have 2 really close,outside and inside of 2-layer window,one on metal-sides shelf,the one outside has LQI ~50-60,others above 100 (city area,not bad,but some wifi networks around,i scan about 5-10 on my phone)

let's take a closer look at the one outside - i attach it to window,it sends few updates(i feel it is working,go mind my own business and after some time it is offline)...so i open the window,pull it off the window,lay it close to coordinator and see blink,nothing...and after ?few seconds? it is online "just now",so again,attach outside,close windows,few updates and dead

the factory one with round display (actually this one: https://github.com/zigpy/zha-device-handlers/issues/2862 ) just works..could be just good signal,i had it lose signal in the past,but it seems more resilient,even disconnecting coordinator it does not care (unlike zigbeetlc firmware,but this would need more testing from me)

so what i think - does it matter slave device(battery powered) does not hear coordinator/hears badly so few packets drop until it is offline and just does not care anymore?is there some setting to listen more/longer period for retransmissions?

and i know my coordinator's antenna sucks,it will end up "if my soldering iron don't kill it,it will make it stronger" anyway :)

pvvx, 2025-09-06:
In TS0201 the ZTU module is not installed according to the [specification](https://developer.tuya.com/en/docs/iot/ztu-module-datasheet?id=Ka45nl4ywgabp):

[image]

As a result, the antenna in TS0201 works very poorly.

Korenchkin, 2025-09-06:
hmmm,thanks for the heads-up,fixed it with pliers..i hope that pins near antenna are unimportant :) (i didn't see any traces there),so i broke it off...also,soldered u.fl on my zb-gw04(from some esp8266,with wire antenna,ouch my eyes hurt),now lqi went from 50-60 to 116 (will see when it is stabilized)

but before breaking outdoor one (now 5 hours offline),it would not send update(under 50cm directly looking at coordinator),1 blink,nothing,second blink,nothing,pressing quickly button several times still nothing...after batteries out and in it sent update...
this device: https://github.com/pvvx/ZigbeeTLc/issues/201

so i don't understand zigbee,why it didn't hear it now,is there some kind of desync?